xyphro/UsbGpib #4: How to compile it?

DaviVolcan, 2020-03-09:
could you describe a litte of the process and the tools that a ineed to compile a testandmeassuarament.bin file? i want the resource name related only to the gpib adress or to each adapter to have a specific name...

Xyp, 2020-03-09:
Hi Davi, 

This is actually very easy to realize. Will add an option to select during runtime the intended behaviour, so either the one we have now (name built by IDN string) or gpib address. Actually assigning a name yourself is doable too without a lot of effort, need to make up my mind here on which of the 2 strategies is better. 

Will furthermore share how to setup a build environment in the evening. 

Best regards, 

Kai

DaviVolcan, 2020-03-10:
Awesome! im really interested in you project because i do think that more people would benefit from it, people can even do more flavors... when a project demands certain features or just by personal preferece... 
your work can reach a lot of people! i do have friends that are ordering pcbs from JLC to build some adapters!

DaviVolcan, 2020-03-10:
Awesome! im really interested in you project because i do think that more people would benefit from it, people can even do more flavors... when a project demands certain features or just by personal preferece... 
your work can reach a lot of people! i do have friends that are ordering pcbs from JLC to build some adapters! 
when how to compile it be clarified... more people can play and benefit! 
Best regards,
Davi

Xyp, 2020-03-10:
Hi!

Due to high workload I'd like to take a pragmatic approach and provide you
my build environment as a first step. I will only manage this weekend to
look into describing it better. And don't want to block you.
Can you let me somehow know your email address? E. G. Mail once to xyohro
at gmail dot com?

Best regards,

Kai

madyn999, 2023-04-19:
Possibly reviving a new thread.  I'm reasonably accomplished with ST Micro's CubeMxIDE.  Used to do a lot of AVR and AVRXMEGA a while back, put that aside to go to STM32 which had more FLASH and RAM.  

I have AVRStudio (Microchip) 7, but it seems a bit of a difficult environment to work in, once one has managed to figure out ST MICRO's eclipse based code.

Did you ever use the AVR Studio tools?  Could you share the setup?  

I'd like to use the adaptor with a TM5006 chassis.  It parallels up to 6 (possible) GPIB controllers.  I'd need to get the software to compile properly before I start to modify it.  What I'd like is to modify the basic code to add up to (say..... 8?) possible GPIB devices.  

If needed, I can easily (it says here) take the USB stack (I think), put it on an ARM processor, let it do a lot of the heavy lifting, and then use I2C to go to the AVR, which may have the appropriate GPIB hardware interfaces.  Ideally, the hardware would support the 6-8 interfaces and the changes would be only software.

I wouldn't mind investigating this, but it's an unfamiliar environment so far.  
Harvey

xyphro, 2023-04-19:
Hi Harvey,

I use the build environment which is provided by the LuFa system which is makefile based.

I think this will help you:
http://www.fourwalledcubicle.com/files/LUFA/Doc/140302/html/_page__getting_started.html

It is not an avr studio IDE based project.

Best regards,

Kai

xyphro, 2023-04-19:
In case you are not used to makefile systems it might be easier to do what you suggest: Use an stm32 and port it and use something like NCA9555 with a i2c level shifter to interface to the 5V domain gpib target. Or build 8 of those circuits and put a usb port on the back side. Gives also the flair of a modern system then :-)

closed 2023-05-03 by xyphro