xyphro/UsbGpib #5: Computer hangs at boot with USB device connected

hppi-johnsson, 2020-08-03:
Hi,

I really appreciate this project. It helped me to create a USBTMC device by removing the GPIB part and implementing own functions.
I got it working in Windows. VISA displays my device and I can communicate with it.

But, if I start the computer with the USBTMC device in the port, the computer simply won't boot. I have tried this on 3 different computers. I suppose it has something to do with the USB enumeration in BIOS. I would be grateful if anyone has an idea what could be wrong. 

The simplest way to reproduce the problem is to comment out lines 382, 387-442 and 454-461 to skip the GPIB part.

caizikun, 2020-08-31:
you can contact  Xyphro by email: Xyphro at gmail.com

Xyp, 2020-08-31:
Hi!

Can you contact me at above email address? Will also post the solution here, once identified. Sorry, only see the issue now first time. 

Best regards, 

Kai

caizikun, 2020-09-01:
johnsson,
you'd better list the programer HW/SW and operation steps. Some error detail may cause problems.

closed 2022-11-24 by xyphro