xyphro/UsbGpib #6: ATMEGA16U4 support

ra1nb0w, 2020-12-06:
Hi,
first of all, thank you very much for sharing this project!
Unfortunately, I soldered an atmega16u4 without re-reading the readme :(
I rebuild `TestAndMeasurement` for that micro and flashed it without bootloader since I have only 16KB of flash (with fuses as suggested). The led works correctly but I don't get the usb descriptor Id and the usb is not recognised. I think that the bootloader set those values but I don't know too much about LUFA. Is there a way to use 16u4 without desoldering it?
thank you

irbsurf, 2021-06-11:
Hi, guess what I did the same except I haven't soldered it yet! Now the correct IC is out of stock! Did you make it work in the end?
Thanks,
Andrew

ra1nb0w, 2021-06-12:
No. At the end I substituted the IC.

closed 2021-06-12 by ra1nb0w

Xyp, 2021-12-02:
It would be possible to Support atmega16U if the bootloader is removed. So you'd loose Update capability to Users that don't have a programmer/skills.
Well, right now everybody needs to bring these tools with him... If I manage to get time I can make something, right now it is unfortunately just extremly limited :-(