xyphro/UsbGpib #8: Python VISA support?

ami3go, 2021-05-18:
Hi. Thanks for sharing it. 
Do you know if it is supported by PyVISA? 
Project looks good. I have some suggestion to make it more robust. 
If this turned out to support python i would like to collaborate improve schematic.

ami3go, 2021-05-18:
I found that python is actually supported. Great!

Xyp, 2021-05-18:
The way it is implemented it is independent on with which software you control it, because it hooks up to visa. Python and matlab are my main preference for automating measurement equipment so what you found is fully correct :-)

ESD... Yeah, can be done. The AVR already has protection diodes internally, it all comes down to the targeted model you want to protect it against.

ami3go, 2021-05-19:
USB protections probably STF202-22T1G or it's kind 
Also when i was thinking to build it myself i saw GPIB bus drivers, like:
SN75ALS160
SN75ALS161
On the  one hand good to have them, on the other probably it is better to keep BOM cheap and use several of devises instead one, if control more then one piece of equipment.

Xyp, 2021-05-19:
Bus drivers I kept indeed on purpose out for the methothology of 1 usb port 1 GPIB device as stuff is cheap. Bus drivers add cost and secondary level complexity. Not saying they are wrong, but with above mindset not required. 

Usb protection can certainly be added. If somebody does not want it, they can simply leave it away.
Esd protections are only as good as the testing behind it. Adding an untested ESD protection is not enabling to make claims on ESD safety.
Adding a pad for a commonly available choice is not hurting at all. 

Best regards,

Kai

ami3go, 2021-05-20:
I will order current version. I think i could test one assembly for ESD as I have access to test lab on work. Let see if need additional protection or not.

Xyp, 2021-05-20:
That would be nice. I can also test agains different models, just in this covid times more effort. Order a few more AVRs to replace after test :-P

closed 2022-11-24 by xyphro