xyphro/UsbGpib #10: Device descriptor request failed

wangwei357, 2021-07-31:
hi, 
I had used your BootLoader.hex to program the microchip. but when i reconnect my computer . there is no disk show in my PC. and the Device Manager will display
 “ Unknown USB device （Device descriptor request failed）” ,What possible causes ？
[image]

xyphro, 2021-07-31:
Hi Wanwei,

Did you program the fuses also as described in the readme.md? This would
cause exactly this issue.

Let me know, curious what the cause is.

Best regards,

Kai

wangwei357, 2021-08-03:
> Hi Wanwei,
> 
> Did you program the fuses also as described in the readme.md? This would
> cause exactly this issue.
> 
> Let me know, curious what the cause is.
> 
> Best regards,
> 
> Kai
> 
> 
> 
> > hi,
> > I had used your BootLoader.hex to program the microchip. but when i
> > reconnect my computer . there is no disk show in my PC. and the Device
> > Manager will display
> > “ Unknown USB device （Device descriptor request failed）” ,What possible
> > causes ？
> > [image: image]
> > <https://user-images.githubusercontent.com/77340375/127742083-d7ecddcd-6d84-4fdf-9309-8f547872fec5.png>

@xyphro  hi  
    It's my fault. it may cause when program the the fuses without usb connected.  
   But  i  had  another issue . when there are multipile gpib driver in gpib network . There is only one driver can discover.  It's  OK for current fwamework?

xyphro, 2022-11-24:
The framework is made to just support a single device.
I think physically it would be possible to drive multiple GPIB instruments, but the software was made with the intention to just support a single one.

closed 2022-11-24 by xyphro