xyphro/UsbGpib #11: TestAndMeasurement.bin don't seem to load

fbdims, 2021-09-03:
Hi there,
When i copy the TestAndMeasurement.bin to the device, and rename it to FLASH.BIN and then reboot, the flash-drive functionality starts again. 

i tried to program both with avrispmkII, and dragon_isp, same result 
Im using linux, but i guess, that should not be an issue... 

any hints ??

Jojeee-xyz, 2022-10-10:
I wasn't expecting this to a be issue with Linux but apparently it is? Just confirmed that using the copy command under a windows terminal works without issue. Device is working now as intended.

xyphro, 2022-10-10:
Hi!

You have to rename it first and then copy it.

Best regards,

Kai

Jojeee-xyz, 2022-10-10:
Hiya,

I tried it both with the original file name and after renaming to 'FLASH.BIN', making sure the file size changed accordingly to the new file. Neither seemed to work, both when using my file manager and the `cp`/`mv` command. 

I think this is a bug in the LUFA bootloader itself, perhaps the same issue [https://www.reddit.com/r/olkb/comments/s0pc6g/flashing_qmk_onto_dz60rgb_ansi_v2/](https://www.reddit.com/r/olkb/comments/s0pc6g/flashing_qmk_onto_dz60rgb_ansi_v2/). A workaround is described using the `dd` command, though I haven't tried it yet:
`dd if=<firmware bin file> of=<path to FLASH.BIN> bs=512 conv=notrunc oflag=direct,sync`

phillid, 2022-11-23:
@Jojeee-xyz thanks - this solved the problem I was having!

I'll submit a pull request in a bit to add this to the docs, since it took me 3 goes at the issues list to find this trick ☺

xyphro, 2022-11-24:
Thanks a lot guys! I have merged the pull request!

closed 2022-12-08 by xyphro