xyphro/UsbGpib #12: Is it possible to port this code to Arduino Pro Micro?

Maxwell3e8, 2021-09-30:
It would be nice to be able to use just the standard Arduino Atmega32u4 board. It looks like just a few GPIB pin assignments need to be moved and interactions with the bootloader?

christakahashi, 2021-12-04:
I'm considering the same thing, mostly because I can't find any 32u4's during this shortage.  You can buy clone pro micro's that are a bit bigger and bring out all the pins.  I assume keeping port D mapped to DIO 1:8 is efficient and remapping the unavailable portD pins would be a bit kludgy.

Xyp, 2021-12-04:
You can also desolder the atmega32u4 from an arduino Board and put it here :-)
Funny enough, I also thought it is hard to get atmega32u4, but bought just läßt week a few from Mouser.

christakahashi, 2021-12-05:
I considered that too, but all the pro micro's I see use the QFN package, which I don't want to fiddle with.

Xyp, 2021-12-05:
But in case you are willing to go there I volunteer to Layout a QFN housings variant.
Or are there any other 5V MCUs good available right now with USB Support?

Maxwell3e8, 2021-12-05:
Personally I don't want to deal with making circuit boards, ordering individual components or installing new IDEs.  A  popular cheap board available on E-bay that can be programmed with Arduino IDE would be much easier to setup, even if it is slightly slower.

Xyp, 2021-12-05:
I understand if this has preferences if you don't want to solder SMD components. You can try porting, it should not be too difficult if there is a good USB stack that you can use for Arduino. I won't do Arduino Support myself as it is already working without it.

christakahashi, 2021-12-05:
I actually laid out a QFN variant last night for assembly in china, since I hate soldering that package.  But they don't have any crystals available.  I asked them if they were going to get a part in stock or can add one, so we'll see.

It's laid out in KiCAD (which has gotten so much better since CERN started adding enhancements).  I'm using the outline imported from your gerbers and the same connector locations so hopefully its compatible with your case.  Assuming I find a crystal, I'm going to order a few and double check my layout is okay and if you want I can add a push request with the layout.

Xyp, 2021-12-06:
Good Action Chris. Looking forward for the pull request!
Did you use the JLC PCB assembly service?

Xyp, 2021-12-06:
I'd really need somebody to show me 1:1 Kicad. What I saw looked so much better to what I knew from years ago.

christakahashi, 2021-12-06:
Yeah I've been using Chinese services for years when selling low volume pcbs.  This will be my first JLC PCB Assembly but their offering has gotten so much better than the last place I used that I have to try it.

Kicad is well worth learning these days especially after autodesk took over eagle.  Most of kicad is very intuitive but its a bit harder coming from eagle since eagle is so weird.

christakahashi, 2021-12-22:
PCBs came in yesterday.  3 days to make two weeks to ship, because I didn't want to pay for good shipping.  Seems to work great!  I'm going to clean up the silk layer a bit to fix some minor "issues"  and send in a pull request.  

KiCAD is on the verge of releasing its 6.0 which has a lot of improvements I love (including fixing some annoying UI oddities).  Among them is (FINALLY!) embedding symbols in the schematic file.  This means that sharing schematics files isn't a path and file versioning nightmare.  I did do it in 5.x originally though, so would you prefer I attempt to package up the 5.x files or just go 6.0.  The tradeoff is that the 5.x files may be hard to import, but KiCAD 6.0 is not technically released, though another feature of 6.0 is that it can coexist with other versions (not previously possible).  I've loving the 6.0 RC so that's what I think is best.  I can add both too but I don't want to make things too confusing.

Xyp, 2021-12-22:
You make me so much wishing to have time to learn Kicad. Maybe one time I'll get that time...

Xyp, 2021-12-22:
Worked with fusion 360 based Eagle and think they completely screwed it up. It got slow and laggy. Forget about the nice New Features if the useability of the core Features is so much impacted!

closed 2022-11-24 by xyphro

dreamcat4, 2023-04-16:
+1 another have here ardino micro (chinese clone version). And just don't want to bother transferring it to another PCB, to have to order those 10x pcbs, be waiting 2 weeks. There is so little on it, to wire up.

If that fails then the can still fall back to ordering the custom PCB.

xyphro, 2023-04-16:
Feel free to port it to arduino. I won't do it, but released all sources.

xyphro, 2023-04-16:
why port btw? You can just flash the hex file and wire it up like shown in the schematic.

dreamcat4, 2023-04-16:
> why port btw? You can just flash the hex file and wire it up like shown in the schematic.

indeed: that is exactly what i am hoping to do. Precisely that

dreamcat4, 2023-04-16:
If to follow your existing guide in the README... what i appear to be missing here is AVRisp programmer. But it seems like it would be possible to make one (from another 2nd arduino)

So it seems anything different will just be about the bootloader stuff, for the flashing process.

xyphro, 2023-04-16:
If all pins are exposed on the board you mentioned it should work. Let me/us know :-)

xyphro, 2023-04-16:
Yes, this project brings its own bootloader with it. Maybe you are also lucky and the arduino BL supports the application image as is? Could be worth a try.