xyphro/UsbGpib #13: Finished product consultation

michaelshow, 2021-10-11:
Hello, xyphro， can your finished product be sold? Can you sell me one?

wgprojects, 2021-12-02:
+1. If you sold kits, I would buy some. It would be really cool if the bootloader came pre-flashed but I guess that might as well be a completed unit. I'd probably buy that too. With your BOM cost at 14, x 3 = $45 price point being reasonable compared to the competition it could be worthwhile. The PyVISA compatibility is huge.

Xyp, 2021-12-02:
I am always considering it :-) The housings would come with some NRE costs that might be an issue. 3D printing is slow and nothing I'd like to do for higher volumes - also in terms of repeatable quality. 
Just a PCB with premounted and preprogrammed Avr and smd components would not be a big issue.

If you send me an email I might help if you don't want to solder yourself: xyphro@gmail.com

closed 2022-11-24 by xyphro