xyphro/UsbGpib #18: Turn off automatic *IDN? readout

larrye11ison, 2022-03-13:
I have several older instruments that don’t support normal SCPI commands like *IDN? so I would like to turn off the automatic *IDN? readout at startup. I'm trying to make the calls to control_in() and write() to turn this off, but I can’t get it to work. 

The *Setting Parameters* section on the main GitHub page has some sample code and says “VM is the opened VISA resource.” I assume that VM is what my code below calls the “instrument” (the “instr” variable). This is based on the following presumptions:

- Your mention of “VISA resource” on the github page matches very well with the return value from the “open_resource()” call.
- The only other thing I could even GUESS could be involved is the pyvisa Resource Manager. But this object does not have methods for control_in() or write(), so it is totally ruled out.

Is there an easy fix for my code below? Should it call control_in() and write() on the “instrument” like this? If so, any idea why it’s not working? Or, if I should be calling these methods on some other object, can you tell me what that is?

```python
import pyvisa
rm = pyvisa.ResourceManager()
all_resources = rm.list_resources()
theAddressToUse = all_resources[1] ## In my case, I want the second item
inst = rm.open_resource(theAddressToUse)
print(inst.query("ID?"))  ## 3488A uses ID?, not *IDN?

## Everything up to this point works fine -- I get the string "HP3488A" returned,
## above, and other commands/queries on the 3488A work as expected.
## But then I try this to try and turn off the automatic *IDN? :
inst.control_in(0xa1, 0x40, 0, 0) 
inst.write('!0001')
## The instrument beeps with an error, I assume because the *IDN? command is being sent yet again...
## But those commands never have any effect on the controller.
```

xyphro, 2022-03-13:
Hi,

I'll make a dedicated python Script and push it to Github + Update you here
once done.

Best regards,

Kai

Xyp, 2022-03-13:
Hi Not Larry :-)

I tried it out and it works. Actually it did not on the first try. This was, because I had an old firmware and this might be your issue here too. This feature was not present in the first firmware version.
Could you download from SW/binaries folder the TestAndMeasurement.bin file and program it as described in the readme.md to your device (short circuit the 2 pins as shown, plug it into the USB port, copy the TestAndMeasurement.bin to Flash.bin and overwrite it)?

I used this sequence to make it work:

import pyvisa as visa

rm = visa.ResourceManager()
r = rm.list_resources()
print(r)
VM = rm.open_resource(r[0]) # Maybe adjust this if your gpibusb adapter is not the 1st returned device
VM.control_in(0xa1, 0x40, 0, 0); # Identificator pulse command (1st step for configuration command
VM.write('!0001') # only sense GPIB address

Let me know if my suspicion with the firmware version was the cause.

BR,

Kai

larrye11ison, 2022-03-13:
Thanks, Kai. BTW, I'm the guy in the US that just received the unit from you a few days ago (after the lengthy shipping time, thanks to the postal workers on both sides of the Atlantic!). 

Because the device was so new, I didn't even consider that it had the older firmware! Turns out that was exactly the problem - I flashed the new TestAndMeasurement.bin firmware and it's working fine now. Of course, now I feel silly for not trying that from the beginning...

closed 2022-03-13 by larrye11ison