xyphro/UsbGpib #23: Version 2 Gerber Files not accepted by PCBway

dazz100, 2023-04-29:
Hi
I really like your project.  It is exactly what I want and need.
I am trying order the PCBs from PCBWay but there may be an issue with the production files for the version 2 pcb.

The files are read by the on-line Gerber viewer found here: [https://www.pcbway.com/project/OnlineGerberViewer.html](url)
If I click on `Get an instant PCB quote`, then upload the Gerber files, they are not accepted.

Can you please check the version 2 files for any errors.

Thanks for making this project public.

xyphro, 2023-04-29:
Hi Dazz,

Each PCB maker has again different requirements in terms of gerber layer
namings. Making it work for all without adjustments is impossible.

I suggest to use jlcpcb or check the layer naming requirements for pcbway
and adjust it accordingly.

Best regards,

Kai

dazz100, 2023-04-30:
Hi
PCBway have now accepted the files without any change.

closed 2023-04-30 by dazz100