xyphro/UsbGpib #26: UsbGpib

zodrax, 2023-06-04:
Hello! Loved your Usb Gpib project. I tried to assemble it, but there were questions ...

I was confused by fuse bits . I tried to emulate the ATmega 32u4 bootloader in Proteus but it gave me an error.

 avrdude -c usbasp -p m32u4 -e -Ulock:w:0x3F:m -Uefuse:w:0xcb:m -Uhfuse:w:0xd8:m -Ulfuse:w:0xff:m avrdude -c usbasp -p m32u4 -U flash:w:BootLoader.hex

xyphro, 2023-06-05:
Hi Zodrax,

Never used Proteus (but it looks interesting).
In terms of fuse bits you basically need to select the xtal osc and disable jtag pins.
However, for the bootloader usage it might be enougj to just select the 16MHz xtal source.

Can proteus simulate USB IP too?

Best regards,

Kai

xyphro, 2023-06-05:
Unintentionally closed this. Reopening agn

closed 2023-06-05 by xyphro