xyphro/UsbGpib #29: GND pour on bottom layer in Kicad design

hobbymachinist, 2023-06-24:
More a question than an issue, is there a reason there's no ground pour on the bottom layer ? Wanted to check first before I try adding one and then check for loops.

Thanks!

xyphro, 2023-06-24:
Hi!

I am on vacation in Japan and don't have access to any PC right now, so
cannot check myself. The Kicad design was provided by another user. I my
eagle design you will find gnd pouring on both sides.

If I'll be home, I'd now quickly add GND pouring, but this will still take
some time.

Best regards,

Kai

hobbymachinist, 2023-06-24:
no worries, it's not urgent. have a good vacation :)

xyphro, 2023-11-01:
BTW @hobbymachinist: Have changed the Kicad file and applied the gnd pouring on both sides.
However: There will be a new Kicad version coming soon (in about 2 weeks) after testing of the design here.

Will close the issue.

closed 2023-11-01 by xyphro