xyphro/UsbGpib #34: sigrok-cli errors

deepfryed, 2023-07-09:
Hi Kai,

I ran into some errors while trying to use sigrok-cli to read the DMM - is there a setting / config option I'm missing ?  FWIW reading and writing to /dev/usbtmc0 works perfectly fine.

https://www.eevblog.com/forum/projects/smuview-a-sigrok-gui-for-power-supplies-multimeters-and-more/msg4950760/#msg4950760

The cli reads measurement fine the first time, but doing it again results in an error. Disconnect and reconnecting, it works again.

```
$ sigrok-cli -d scpi-dmm:conn=usbtmc/03eb.2065 --samples 5 -c measured_quantity=voltage/dc
P1: 4.99998 V DC
P1: 4.99998 V DC
P1: 4.99998 V DC
P1: 4.99998 V DC
P1: 4.99998 V DC
...

```
$ sigrok-cli -d scpi-dmm:conn=usbtmc/03eb.2065 --samples 5 -c measured_quantity=voltage/dc
sr: scpi_usbtmc: USBTMC bulk in transfer error: LIBUSB_ERROR_TIMEOUT.
sr: scpi_usbtmc: USBTMC bulk out transfer error: LIBUSB_ERROR_TIMEOUT.
sr: scpi_usbtmc: USBTMC bulk out transfer error: LIBUSB_ERROR_TIMEOUT.
sr: scpi_usbtmc: USBTMC bulk out transfer error: LIBUSB_ERROR_TIMEOUT.
sr: scpi_usbtmc: USBTMC bulk out transfer error: LIBUSB_ERROR_TIMEOUT.
sr: scpi_usbtmc: USBTMC bulk out transfer error: LIBUSB_ERROR_TIMEOUT.
sr: scpi_usbtmc: USBTMC bulk out transfer error: LIBUSB_ERROR_TIMEOUT.
sr: scpi_usbtmc: USBTMC bulk out transfer error: LIBUSB_ERROR_TIMEOUT.
```

xyphro, 2023-10-15:
I can only guess here. How does sigrok know which multimeter type it is and how to control it?
There are many different multimeters with many different ways to control them. There is also no common auto identify method present, especially old multimeters don't even support *idn?

xyphro, 2023-10-15:
You could try to make a usb trace and see how it controls it. Or a visa trace if possible for your OS.

closed 2023-12-31 by xyphro