xyphro/UsbGpib #35: Changing timeout.

mariusCZ, 2023-07-25:
Hi,

I started having an issue where changing the timeout in my LabView program did not actually change the timeout of the device read function. I took a look at the code and saw that there is a timeout incorporated in to the adapter's code. Where can I change the default timeout value for a reading operation? Thanks!

xyphro, 2023-07-25:
The timeouts are set on pc side by sending device.timeout = 10000 (just as
an example using python). I am no labview expert, but there also has to be
a way to set timeouts towards visa domain.

I have taken care to implement timeouts very well, this was in fact one of
the more complex things during the development: not just get it
functionally work, but also let it behave well on errors.

There is no need to change firmware!

I tested this on slow actions like reading very long waveform data from
device or hardcopies.

mariusCZ, 2023-07-25:
That's quite interesting. Is there perhaps some minimum allowed timeout setting?

There is a way to set a timeout with VISA in Labview, however, setting it to something like 100 ms doesn't change anything, as in, the timeout still takes a second or two.

xyphro, 2023-07-25:
Interesting. From firmware perspective, I did not set a timeout limit.

BTW: Why do you intentionally want to run in a timeout? Are you trying to
see if the device is still busy with an operation and poll that quick?

mariusCZ, 2023-07-25:
Sorry for having bothered you, it was an issue on my end! I was running Labview in a virtual machine and while communication seemingly worked fine, for some reason the timeout was not being sent properly. Now it works when I run Labview natively under Linux!

The program I am using is quite old and has some weird practices. Basically, both the write and read are included in a single sub-vi (in a single function if talking in terms of normal programming). So I use the short timeout to determine whether the command I have sent returns any type of data. Other methods have proven to be unreliable and to try and overhaul this system would be, quite frankly, a lot of effort that is not worth to put in.

Thank you for the help!

closed 2023-07-25 by mariusCZ