xyphro/UsbGpib #37: someting went wrong with "Setting Parameters", I think.

bnzf, 2023-09-05:
Hi,
I tried the "Setting Parameters" for read termination and automatic instrument ID. In the first attempt something seemed to go wrong (I think it was caused by that I tried to send both commands at once). Then I repeated the procedure command by command. All looked fine.

But after that rm.list_resources() does not work anymore, nothing happens, I get no Python prompt afterwards.
I rebooted the UsbGpib, I disconnected it, started a new python session, I reinstalled the firmware,... no changes. However the instrument is well visible with NI-Max and data coming from the instrument is there visible.

As I cannot connect via pyvisa anymore, I can not repeat or clear the parameter setting from there.
Is there a possibility to reset the  EEProm somehow without pyvisa?
Or can I use the NI-Max tool for sending commands to (re)set the EEProm?

I use python from an OS-X machine. NI-Max is installed on a Windows-7 machine. The GPIB instrument is a HP5370 in talk only mode (it does not support the automatic instrument ID and has \r\n termination).

xyphro, 2023-09-05:
read termination setting does influence the receiving of data from the
instrument which is only happening when you read.

I recently had issues too when I installed  tek openview software that
under the hood installed tekvisa which got my pyvisa into a non working
state (independent of gpib, any visa stuff stopped working). Deinstalling
tekvisa made it working again.

I somehow think that it might be some general visa issue. Or maybe you have
from other tests a libusb driver installed instead of a generic usbtmc
driver?
Do you have other instruments that you can crosscheck on your mac os PC to
see if visa works?

I am not too familiar with mac os. Does it have an equivalent to device
manager to see which driver is used for usbgpib device?
Or is lsusb or so possible?

bnzf, 2023-09-06:
I tested an instrument with native USBTMC. On OSX there also rm.list_resources() stalled. I then uninstalled NI-visa (a lot of work as they provide no proper uninstaller but only instructions what to delete) and reinstalled it, after that all worked fine again. I however did not install anything before  I got in trouble with pyvisa, I only messed around with the "Setting Parameters" commands.

I then did turn off the automatic instrument ID  as described in the instructions. This worked, I no longer get an error message on the instrument when turning the USBGpib adapter on.

However the setting of the read termination does not work entirely as expected. 
It works in the sense that inst.read() now works. But I get random new lines in the output:

```
>>> while True:
...     print(inst.read())
... 
TI = 1.00290000000E-07

TI = 1.00210000000E-07
TI = 1.00270000000E-07

TI = 1.00230000000E-07

TI = 1.00270000000E-07
TI = 1.00290000000E-07
TI = 1.00210000000E-07
```
When I do inst.read_bytes() I get:

```
>>> print(inst.read_bytes(48))                                                  
b'TI = 1.00390000000E-07\rTI = 1.00310000000E-07\r\nT'
>>> 

```

So the first \n is missing, but is there again in the second set of data. 
I would have expected to see here no \r and \n anymore.

Without the read termination setting to \r, inst.read_bytes() never dropped a \r or \n, so it is not caused by the data send from the instrument.

bnzf, 2023-09-06:
P.S. It looks that I get the random new lines only when the Instrument is set in "listen+talk"-mode. If I set the switch on the instrument to talk-only there are no additional new lines.

xyphro, 2023-10-17:
I will try to reproduce it this evening.

xyphro, 2023-12-31:
I could btw never reproduce it. Does it still occur with latest release candidate?

bnzf, 2024-01-02:
I tried now the release candidate software.

With the adapter where I applied the "Setting Parameters" modification, I get in talk only mode, still, the desired output. Diagnostic with inst.read_bytes() produces:

print(inst.read_bytes(96))   
b' 9.99324000000E+06\rFRQ= 9.99909000000E+06\rFRQ= 9.99714000000E+06\rFRQ= 9.99714000000E+06\rFRQ= 9.9'

In "listen+talk"-mode the random behavior is gone, but I get an extra empty line in the inst.read() loop. Diagnostic with inst.read_bytes() produces:

print(inst.read_bytes(96))  
b'\nFRQ= 9.99324000000E+06\r\nFRQ= 1.00030000000E+07\r\nFRQ= 9.99519000000E+06\r\nFRQ= 9.99519000000E+06\r'

On an other of your adapters with the release candidate software in default state, i.e. without the "Setting Parameters" for read termination and automatic instrument ID, I get in both settings of the instrument (talk only, listen+talk) the same:

print(inst.read_bytes(96))
b'RQ= 9.99887000000E+06\r\nFRQ= 9.99497000000E+06\r\nFRQ= 9.99692000000E+06\r\nFRQ= 9.99497000000E+06\r\nF'

here I defintively need to use inst.read_bytes() here as the normal loop with inst.read() misses the termination (of cause) and runs in a timeout error.

P.S. In the README.md the instructions for putting the binaries on the adapter have disappeared, I had to look in history on how to do.

bnzf, 2024-01-02:
PPS. The HP5370B manual says it uses IEEE Standard 488-1975, so the very first 488.

xyphro, 2024-01-02:
The sudden appearance of the /n in talk only vs. Listen+talk is a miracle to me.

Would you volunteer to help with a logic analyzer trace if I tell you what signals to monitor (ideally 16 channel, 8 would also work)? Of course only if you have one... in case you are in europe I could borrow/ship you a 16 channel fx2 based analyzer which works with sigrok.

xyphro, 2024-01-12:
Short update: Have just packed the shipment of the 16 channel logic analyzer + cable and will bring it to postbox tomorrow.

xyphro, 2024-01-13:
Today a friend came by with a HP5535a as a present. Not exactly the instrument that you have, but likely similar when it comes to age.
While you wait for the LA to arrive, I will check if I see similar behavior with that one!

[image]

xyphro, 2024-01-21:
...let me update here. We debugged the issue remotely and the cause is clear.

The instrument returns messages in the format:
`FRQ= 9.99999993500E+06<CR><LF>`

The GPIB read termination was set to <CR>. This resulted in messages returned in this format:

`<LF>FRQ= 9.99999993500E+06<CR>`

the print() will add also another termination.

The solution was:
1. configure gpib read termination as LF termination:
dev.control_in(0xa1, 0x40, 0, 0, 1); # set read termination to LF
dev.write('!term lf')
2. python can strip away read termination automatically, if it is correctly configured. So configuring it in this case like this strips it automagically away:
`dev.read_termination = '\r\n'`

Already with the 1st mentioned item the behavior in talk only and listen+talk mode was identical and no random inserted \r or \n were seen.

xyphro, 2024-01-21:
Closing the issue.

closed 2024-01-21 by xyphro