xyphro/UsbGpib #39: Power-on error

htduan, 2023-09-14:
If the GPIBUSB adapter is connected to a powered-off instrument while itself is connected to USB (or simply powered by USB), powering on the instrument will turn on "error" annunciator. Subsequent SCPI query on the error info returns:

-113,"Undefined header"
-420,"Query UNTERMINATED"

Tested on Agilent 34401A and E3632A, and such error persists. Both instruments perform self-test right after power-on, and the issue might be related to the partially received *IDN? command sent by the adapter during the self-test period.

xyphro, 2023-09-14:
Check the section "Automatic instrument identification readout in the
readme.md to turn that feature off, then you won't get this issue anymore.

xyphro, 2023-10-15:
I close this issue as it likely got solved for you.

closed 2023-10-15 by xyphro