xyphro/UsbGpib #40: Does it work with harddrives?

JPT77, 2023-10-04:
Hi, 

I've got an HP9000/322 plus drives. 

will I be able to read the disks from PC?
will I be able to emulate a harddrive from PC?

thanks

xyphro, 2023-10-04:
Very good question, never tried a Harddrive.
If it is following what normal measurement equipments have in terms of GPI
communication it should work.
Is there any information available in terms of communication documentation
for those harddrives?

Emulating will likely not work, as the USBGPIB adapter only works as a
controller, but not as a device. HW wise it would be possible, but it is
not implemented in SW. It would also not work easily together with USBTMC
protocol.

Best regards,

Kai

JPT77, 2023-10-05:
Well, here is an overview of the hard drive protocols:
https://www.hp9845.net/9845/projects/hpdrive/
You do understand far more than me, eh?

I expect it to be a LOT of work to make your adapter compatible. 
maybe there is a way to join forces? 
an USB adapter could be driven by a Raspberry PI... So one would not need a full PC for emulating a drive.

xyphro, 2023-10-15:
Hi!

I think what you mostly need is a SW layer on top of usbtmc. Another method would be to make a usbmass storage based firmware recycling the gpib routines from this project.
I am afraid that this is outside of the project scope.

Best regards,

Kai

closed 2023-10-15 by xyphro