xyphro/UsbGpib #42: KE5FX GPIB Toolkit help

crimus, 2023-10-23:
Any advice to make it been recognized by KE5FX GPIB Toolkit?
I tried playing without success with connect.ini.

Any advice?

xyphro, 2023-10-23:
Hi!

It seems that this one is focussed on gpib and not using generic visa strings like usbtmc. It might just work with devices that expose as gpib:: in the ressource string.
But just had a quick look.

The easiest might be to contact the autor of this toolkit to ask if he can allow also usb ressource string. In case it relies on visa API it should be easy for him. In case he builds up on older 488 libraries it will be hard for him.

Longer story: i worked on making a visa.dll that acts as interposer to a normal visa dll to be able to change device ressource strings automatically. It worked nice on my pc but not on others and stopped it due to significant efforts required. It is also a very hacky solution.

Best regards,

Kai

crimus, 2023-10-23:
Actually, KE5FX Toolkit works on various interfaces, but i cannot be sure if they rely all on GPIB:: interface.
Check http://www.ke5fx.com/gpib/faq.htm

Anyway, thanks for your effort in doing/sharing the inteface and some help on this particular issue.

Regards,
Cristiano

caiser01, 2023-10-23:
The KE5FX software does not use VISA and only supports adapters that present GPIB:: interfaces or Prologix-compatible adapters. Unless KE5FX's author adds support for USB:: interfaces, that software will not work with the xyphro adapter.

xyphro, 2023-10-23:
Indeex, i cannot change the usbx:: prefix as this is a usbtmc device.

A dll that exposes visa functions and then calls the real visa dll functions is the only (hacky) way I see now if author does not support generic visa names. The dll would pass through all vis functions 1:1, but just change the visa ressource name.

crimus, 2023-10-23:
Thanks for the informations provided.

I hope this is useful for someone else.

Cristiano

xyphro, 2023-12-31:
I close it for now. If more users demand a man in the middle visa approach to convert GPIB:: strings to USB:: ones to supoort old software let me know. I have a rudimentary forwarder dll made for it but it requires more effort, especially a GUI for easy installation and cobfiguration.
Maybe somebody volunteers also to take that effort? :-)

closed 2023-12-31 by xyphro