xyphro/UsbGpib #43: Dos not apear as usb

yngve-hansen, 2023-10-26:
After setting fuses i cant get it to show up. Guess something went wrong but cant understand what.
[image]

xyphro, 2023-10-26:
Os ot possible that you already have the .bin file programmed too?
Maybe before you had set the fuses?

If so: it is expected defined behavior. It only appears on usb side when it is connected to an instrument and the instrument is powered up.

yngve-hansen, 2023-10-27:
Not so lucky, it dos not show up.

xyphro, 2023-10-27:
> Not so lucky, it dos not show up.
> 

Can you explain a bit more?
Did it show up before?

What were your exact stepy?

yngve-hansen, 2023-10-27:
This is a new module, just initiuliced it and one more.
Same problem on bouth, Crystal is 16mhz.

xyphro, 2023-10-27:
> This is a new module, just initiuliced it and one more.
> Same problem on bouth, Crystal is 16mhz.
> still responsive on isp.

Do you use the board design I shared or is it some arduino module or so?

yngve-hansen, 2023-10-27:
It is your design nr w588420as1f1.

yngve-hansen, 2023-10-27:
I have 5 boards, orderd new. Same fault on the 2 first, will trye a 3 now, but dos not have high hopes.

[image]
[image]

xyphro, 2023-10-27:
w588420as1f1 is a number your PCB production company has added. JLCPCB does this for example.
You used the Kicad project files, which is a user contributed design. But it was reported to work.

The picture seems a bit blurry, C1 looks a bit misplaced. This will likely not be the reason.
Did you solder the IC with paste? So is the Thermal pad (=exposed die pad) connected?
Are the resistors R2 and R3 22 Ohm in value (maybe better measure it once with a multimeter (with USB cable disconnected)?
Does your PC detect the USB device in any way? If you use windows: Does it make any sound when you connect it?

Can you also check once if the XTAL oscillates really on 16 MHz when you connect it to USB? by probing with a scope? Not that it is a harmonic type or so.
You can measure it here (ensure to also have GND connected. You can also connect the aligator clip to the GPIB connector screws or so (better would be close to XTAL).
[image]

Fuses are a NIGHTMARE on the AVR controllers. Each programming tool has again a different definition (inverted / non inverted). Atmel should have defined a scheme where the fuses are part of the .HEX file... However, I just crosschecked the fuses again that you used and they are correct.

You can also try to flash this .hex file to see if it changes something: This is an image, where the bootloader and the UsbGpib application are combined in a single image. Note: This one will really only be visible to the PC when you really have it connected to a GPIB instrument and the GPIB instrument is powered on.
[GpibUsbCombined.zip](https://github.com/xyphro/UsbGpib/files/13192901/GpibUsbCombined.zip)

yngve-hansen, 2023-10-28:
r2 and r3 is 22 ohm, the crystal works(sinus) not tested frekvens.
Smd is solderd by JLCPCB connectors by me.
Windows dos not registrate it in any way.
Found the first problem, it is one command for the fuses and one for boot. They must be run seperatly.
Now it apear in System but missing driver. Will continue to morrow.

xyphro, 2023-10-28:
Install a visa like R&S visa:
https://www.rohde-schwarz.com/de/applikationen/r-s-visa-application-note_56280-148812.html

This one includes usbtmc driver.

xyphro, 2023-10-30:
Hvis den vises som en stasjon, betyr det at selve GPIB-applikasjonen fortsatt må blinke.  Hvordan det fungerer er beskrevet i readme.md.  Bare skriv inn copy testandmeasurement.bin F:\flash.bin på kommandolinjen.  F-stasjonen må åpenbart byttes ut med det GPIB-adapteren har.

yngve-hansen, 2023-10-30:
Mulig den ikke fult ut registrerer hp 54201a, registrerer vertfall ikke fluke 45 i det heletatt.
I enhetsbehandling dukker hp 5421a opp når det slås på.
[image]
[image]

xyphro, 2023-10-30:
Hei Yngve, det ser veldig bra ut. Det er normalt at USBTMC-enheter ikke vises etter navn i enhetsbehandlingen.
Men hvis du søker etter enheter i R&S Visa med testerapplikasjonen, vil de bli oppført der.

Alternativt kan du vise USBTMC-enheter med python og Pyvisa:

`import pyvisa
rm = pyvisa.ResourceManager()
print(rm.list_resources())
`
hvis en feil vises når du kaller import pyvisa-linjen, må pyvisa-pakken fortsatt installeres av:
`pip install pyvisa` samtaler

xyphro, 2023-10-30:
Her er et eksempel på hvordan det ser ut i enhetsbehandlingen min:
[image]

i NI-MAX vises den som en usbtmc-ressurs (jeg kan ikke bruke R&S Visa akkurat nå)
[image]

Og i Python ser det slik ut:
[image]

yngve-hansen, 2023-11-04:
Finner ikke drivere til mine instrumenter, mulig det er årsaken til at den ikke dukker opp i  filutforsker.
En annen mulighet er at det har med win11, men kan ikke bruke mere tid på dette nå.
Takker for support selv om det ble resultatløst.

closed 2024-01-13 by xyphro