xyphro/UsbGpib #44: When will V2 release? :)

galaris, 2023-11-17:
Can't wait for early holidays ;D

xyphro, 2023-11-17:
Haha, stay tuned!

I made a stupid issue: i swapped odd and even pins so the connector has to mount on the wrong side, but: it works!

New pcbs are ordered, likely to arrive next week. Then I will test them immediately and release! Don't want to create a mess by releasing untested stuff.

Housing design is already finished. Busy printing 50 of them - takes multiple days :-(

Btw: i started working on speed upgrade. By a few risk free changes I got it from 85kB/s peak for writes to 400kBytes/s with room for more increase.
Reads are likely more important (waveform/screenshot downloads), those I will benchmark and improve also, likely significantly.

Never looked into speed before, but it is time for it! And I will also see how fast NI USB adapter is as comparison. Have built a dummy device doing the 3 wire handshake with logic ICs, so that pushback from the instrument does not influence the measurement.

xyphro, 2023-11-17:
If you cannot wait, I could drop you here already the stls of the new housing :-)

xyphro, 2023-11-20:
Update: Arrival of the fixed REV 2 board variant is scheduled for wednesday. In case there are no issues at customs.

xyphro, 2023-11-21:
Another update: received 50 pieces already today and will test this evening :-)

xyphro, 2023-11-21:
Test: succeeded, also with upcoming firmware.
Now i need some time to update git structure for the different hw revisions and documentation for it.

In case you cannot wait, send me a mail to xyphro@gmail.com and I'll mail it to you.

Best regards,

Kai

makeitsquare, 2023-12-14:
Hi,
awesome project! Thanks a lot for sharing! Are the Gerber files in HW/Gerber Files V2 for the Rev2 boards? I was expecting to see a rectangular PCB with a USB-C connector, but the V2 files are still for a L shaped PCB with a normal USB connector. So I guess Rev 2 != V2?
Thanks again,
Markus

xyphro, 2023-12-14:
The v2 is not yet released. I have however sent it already to a few users
and proven that it works. I am very busy on the sw side and did not have
time to restructure the site and write instructions yet.

xyphro, 2023-12-19:
It is done, the new hw and a sw release candidate is released (hooray ;-))

xyphro, 2023-12-19:
I have opened a discussion section and posted about the release here:
https://github.com/xyphro/UsbGpib/discussions

Any feedback is highly valued- on HW & on new SW release candidate.

For now I close this thread - you can add to the discussions section.

closed 2023-12-19 by xyphro