xyphro/UsbGpib #53: usbgpib_conf.py a quick hack using python-usbtmc

jimthouston, 2024-01-15:
I have been playing with UsbGpib for a couple months.  I'm a Linux user and have been using pyvisa with the pyvisa-py back end.  This doesn't support the control_in method.

I found that python-usbtmc works nicely for sending the config options.

$ cat usbgpib_conf.py
```
import sys
import usbtmc

inst = usbtmc.Instrument(0x3eb, 0x2065)
if len(sys.argv) < 2:
    print('Usage: usbgpib_conf command')
    for cmd in [ '!autoid?', '!term?', '!string?', '!ver?' ]:
        inst.pulse()
        print(cmd, inst.ask(cmd))
    sys.exit(0)

cmd = '!' + ' '.join(sys.argv[1:])
print(cmd)
inst.pulse()
if cmd[-1] == '?':
    print(inst.ask(cmd))
else:
    inst.write(cmd)
```

$ python3 usbgpib_conf.py
Usage: usbgpib_conf command
!autoid? off
!term?  eoi
!string? normal
!ver? V1.6

$ python3 usbgpib_conf.py autoid on

rjordans, 2024-01-15:
Ah, good to see this here.  I was just looking into the same issue with pyvisa-py limitations here

jimthouston, 2024-01-15:
The python-usbtmc is on github.  The readme has information for use on both Linux and Windows.

I imagine that you will add  a sub-directory for scripts like this.  I was thinking that this was a workaround for pyvisa-py not implementing control_in.  It probably needs magic to select which UsbGpib device to talk to if there is more than one.

xyphro, 2024-01-15:
Yeah, it is just hard to structure this in a proper way. Just having a folder and dropping randon scripts in it is a bit difficult to understand. A post under discussion is maybe the best. It will never get closed, users can modify, comment and give feedback.

Modification by users is also possible of course in the repo itself, but in practice over the last years there were maybe 3 pull requests - so it rarely happens.

xyphro, 2024-01-16:
Hi Jim,

will close this ticket now.
I have preserved your great script in the discussion section: https://github.com/xyphro/UsbGpib/discussions/55

Thanks a lot for sharing this!

Best regards,

Kai

closed 2024-01-16 by xyphro