xyphro/UsbGpib #54: testing with Tektronix DSA 602A

jimthouston, 2024-01-15:
I'm using Linux, pyvisa and pyvisa-py.  I have had good luck testing with an HP-3478A and HP6642A.  Now testing with the DSA602A I'm seeing some odd behavior including a  consistent hang when the python script exits.

$ cat t1.py
`import pyvisa
rm = pyvisa.ResourceManager()
print(rm.list_resources())
`
$ python3 t1.py
('USB1::0x03eb::0x2065::GPIB_01_34231323231351E04071',)
^C
$

I captured a system call trace and the last system call before I use '^C'
is:
futex(0x7f939c5ee664, FUTEX_WAIT_PRIVATE, 12, NULL) = ? ERESTARTSYS

I also tried using the "-m pdb" debugger option.  I had hoped that it would drop me in the debugger when I interrupted the hang.  Instead it puts me into the debugger when the program exits and would run the program again when I continue.

Before I updated to the latest UsbGpib firmware I was also having a problem with my Lenovo T470 laptop hanging on power on with the UsbGpib connected to the DSA602A.  It got stuck displaying "to interrupt the normal boot sequence, hit Enter".  I could boot if I disconnected either the USB or the DSA 602A.  I have not seen this problem with the new firmware.

xyphro, 2024-01-15:
Hi Jim,

not sure how that can be related to the adapter itself, but I don't want to say yet that I am 100% confident on this statement - so let's investigate!
But to explain: There is 0 interaction on GPIB side when you do an instrument scan, this is why I think a bit that it is not dependent on the adapter.

Is t1.py really only consisting out of the 3 commands?
What I see is that the print seems finalized cause the bracket closes. 
Can you maybe add a 4th line with a print?

Does your Visa tool of choice list the instrument properly?
Can you tell me a bit more, like e.g. which visa provider you use? 
Is this using pyvisa-py? I see you use Linux, right?

A capture with wireshark and usbpcap while running your script could help a lot to trace this down.

ERESTARTSYS does not neccesarily point to a cause here unfortunately.

Best regards,

Kai

xyphro, 2024-01-15:
BTW: you are not saying it explicitely... But what happens when you run the script without the adapter beeing attached?

xyphro, 2024-01-15:
BTW: Pyvisa-py had quite some issues in the past on USBTMC and the implementation is very basic. I would not use it anymore after having run into issues, but can of course understand if you do this...

R&S Visa is an excellent choice also for Linux and works nice with USB. It is also not that bloated as other options...

jimthouston, 2024-01-15:
I'm sorry I probably should have done more testing before posting.
I switched computers when I started testing with the DSA602A.  I just tried connecting the second computer to the HP3478A and I have the hang with the HP3478A  and also with the HP3478A powered off.  I wasn't having the problem with the pyvisa and pyvisa-py on the first computer.

So I agree that this is likely to be a pyvisa/pyvisa-py bug not a UsbGpib problem.  If you can leave this open for a while I expect that I will track down the issue eventually.

xyphro, 2024-01-15:
Thanks for the update Jim, i will leave it indeed open until we have full confidence! Curious what your investigation will show.

jimthouston, 2024-01-16:
I used pyvisa-info on both systems and found that I had installed the librevisa library on the system which has the hang problem.
The dpkg.log shows I did this in 2022 but I don't remember.

I tried using pyvisa.ResourceManager('@py') to force the use of pyvisa-py and it doesn't hang.  So the problem is with the librevisa library.

xyphro, 2024-01-16:
Thanks for the update Jim! So it is unrelated to the GPIB Adapter.
In case you see anything else popping up, please report again!

closed 2024-01-16 by xyphro