xyphro/UsbGpib #59: query the string length setting always returns "normal"

mt1970, 2024-03-17:
Hi,

With parameter setting '!string short' , the query '!string?' always returns "normal".

Maurice

xyphro, 2024-03-17:
Hi Maurice,

thanks for reporting, I will check during this week.

Best regards,

Kai

xyphro, 2024-03-17:
[TestAndMeasurement_Dazz&Maurice.zip](https://github.com/xyphro/UsbGpib/files/14628407/TestAndMeasurement_Dazz.Maurice.zip)

Hi Maurice,

this one was easy to spot.
It is fixed - can you try?

Best regards,

Kai

mt1970, 2024-03-17:
Hello Kai,
It works.
Thanks for the fast update.

Maurice

xyphro, 2024-03-17:
Welcome, thanks for trying!

I will commit this firmware update once the other issue which Dazz reported is sorted out.

dazz100, 2024-03-18:
Hi

I can now repeat the test script without issue.  So the test firmware has fixed that problem.
I think this fix has revealed or created a similar issue. 

I plugged the adapter into the usb. I then switched on the emc-30 and ran the script.   Running the script put the emc30 into remote mode.  The following output is normal and correct.   

```
D:\darren\My Documents\Electronics\Electro-Metrics EMC-30\GPIB_Test>python EMC30_gpib_rd_STATUS.py
###### Simple GPIB test for emc30
# Executes read STATUS

# Show the list of Resources
('USB0::0x03EB::0x2065::GPIB_15_24238323232351C01171::INSTR', 'ASRL1::INSTR')
# Opening the emc30

Set Auto ID = off
Set Line Feed = on
Save settings to eeprom
Read emc-30 status
55
```
The following output is not correct.  This occurred after I front panel switched to local mode.
 

```
D:\darren\My Documents\Electronics\Electro-Metrics EMC-30\GPIB_Test>python EMC30_gpib_rd_STATUS.py
###### Simple GPIB test for emc30
# Executes read STATUS

# Show the list of Resources
('ASRL1::INSTR',)
# Opening the emc30
Traceback (most recent call last):
  File "D:\darren\My Documents\Electronics\Electro-Metrics EMC-30\GPIB_Test\EMC30_gpib_rd_STATUS.py", line 30, in <module>
    emc30 = rm.open_resource('USB0::0x03EB::0x2065::GPIB_15_24238323232351C01171::INSTR', send_end=False,)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "G:\Programs\Python310\Lib\site-packages\pyvisa\highlevel.py", line 3292, in open_resource
    res.open(access_mode, open_timeout)
  File "G:\Programs\Python310\Lib\site-packages\pyvisa\resources\resource.py", line 281, in open
    self.session, status = self._resource_manager.open_bare_resource(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "G:\Programs\Python310\Lib\site-packages\pyvisa\highlevel.py", line 3217, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "G:\Programs\Python310\Lib\site-packages\pyvisa\ctwrapper\functions.py", line 1850, in open
    ret = library.viOpen(
          ^^^^^^^^^^^^^^^
  File "G:\Programs\Python310\Lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "G:\Programs\Python310\Lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.
```
Unplugging/plugging (cycling power) the adapter usb does not restore normal operation.

Running the test script with the adapter connected to the usb and the adapter plugged into the emc-30 gpib bus, but with the emc-30 switched off, produces the same faulty output.  The output errors long before the script reaches the code where a command is sent to the emc-30.

When I start with power switched off on the emc-30 and the adapter usb disconnected, I can no longer get normal output. 

My hypothesis is that saving the settings in the adapter somehow causes an issue.

My next test is to reprogram the adapter firmware and run the script again.  My aim is to be able to reproduce the issue.

dazz100, 2024-03-18:
Hi
I reprogrammed the firmware in the adapter, then repeated the actions that produce correct output.
I plugged the adapter into the OFF emc-30.
I plugged the adapter into the usb port.
I powered the emc 30 and did not touch and front panel controls.
I ran the script, which produced errored output.

```
D:\darren\My Documents\Electronics\Electro-Metrics EMC-30\GPIB_Test>python EMC30_gpib_rd_STATUS.py
###### Simple GPIB test for emc30
# Executes read STATUS

# Show the list of Resources
('ASRL1::INSTR',)
# Opening the emc30
Traceback (most recent call last):
  File "D:\Electro-Metrics EMC-30\GPIB_Test\EMC30_gpib_rd_STATUS.py", line 30, in <module>
    emc30 = rm.open_resource('USB0::0x03EB::0x2065::GPIB_15_24238323232351C01171::INSTR', send_end=False,)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "G:\Programs\Python310\Lib\site-packages\pyvisa\highlevel.py", line 3292, in open_resource
    res.open(access_mode, open_timeout)
  File "G:\Programs\Python310\Lib\site-packages\pyvisa\resources\resource.py", line 281, in open
    self.session, status = self._resource_manager.open_bare_resource(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "G:\Programs\Python310\Lib\site-packages\pyvisa\highlevel.py", line 3217, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "G:\Programs\Python310\Lib\site-packages\pyvisa\ctwrapper\functions.py", line 1850, in open
    ret = library.viOpen(
          ^^^^^^^^^^^^^^^
  File "G:\Programs\Python310\Lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "G:\Programs\Python310\Lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.
```

I noticed an error in the output of AVRdude:

```
G:\Programs\avrdude-v7.1\test_20240318>..\avrdude -c usbasp -p m32u4 -e -Ulock:w:0x3F:m -Uefuse:w:0xcb:m -Uhfuse:w:0xd8:m -Ulfuse:w:0xff:m -U flash:w:TestAndMeasurement.bin -U flash:w:BootLoader.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9587 (probably m32u4)
avrdude: erasing chip
avrdude: reading input file 0x3F for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
 ***failed;
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0x3F
avrdude warning: ignoring mismatch in unused bits of lock
        (device 0xff != input 0x3f); to prevent this warning set
        unused bits to 1 when writing (double check with datasheet)
avrdude: 1 byte of lock verified
avrdude: reading input file 0xcb for efuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte efuse ...
 ***failed;
avrdude: 1 byte of efuse written
avrdude: verifying efuse memory against 0xcb
avrdude warning: ignoring mismatch in unused bits of efuse
        (device 0xfb != input 0xcb); to prevent this warning set
        unused bits to 1 when writing (double check with datasheet)
avrdude: 1 byte of efuse verified
avrdude: reading input file 0xd8 for hfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte hfuse ...
avrdude: 1 byte of hfuse written
avrdude: verifying hfuse memory against 0xd8
avrdude: 1 byte of hfuse verified
avrdude: reading input file 0xff for lfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lfuse ...
avrdude: 1 byte of lfuse written
avrdude: verifying lfuse memory against 0xff
avrdude: 1 byte of lfuse verified
avrdude: reading input file TestAndMeasurement.bin for flash
         with 13718 bytes in 1 section within [0, 0x3595]
         using 108 pages and 106 pad bytes
avrdude: writing 13718 bytes flash ...

Writing | ################################################## | 100% 6.87 s

avrdude: 13718 bytes of flash written
avrdude: verifying flash memory against TestAndMeasurement.bin

Reading | ################################################## | 100% 8.43 s

avrdude: 13718 bytes of flash verified
avrdude: reading input file BootLoader.hex for flash
         with 5724 bytes in 5 sections within [0x6000, 0x7fff]
         using 46 pages and 164 pad bytes
avrdude: writing 5724 bytes flash ...

Writing | ################################################## | 100% 0.15 s

avrdude: 5724 bytes of flash written
avrdude: verifying flash memory against BootLoader.hex

Reading | ################################################## | 100% 0.02 s

avrdude: 5724 bytes of flash verified

avrdude done.  Thank you.

```

xyphro, 2025-01-11:
I am committing the firware binary and source in a few minutes containing the fix.

closed 2025-01-11 by xyphro