xyphro/UsbGpib #63: Version 2, Two LED

dazz100, 2024-09-01:
Hi
I have just finished assembly of my USB-GPIB adapters.  I have just noticed your ver 2 design has two LEDs.   You have used the same pins as my version of your design.  That is good news for me.
You have also used pin 41 to sense 2 LEDs which makes perfect sense.  I don't have that on my version, but I have figured out how to retrofit the connection to gnd to be 100% compatible with your version 2.

I have just downloaded the latest version of firmware with the intention of modifying it for 2x LEDs.  

Have you already modified the firmware to utilize the two LEDs?  If so, what do they indicate?

dazz100, 2024-10-23:
Hi
I have finished writing code for the 2x LED version 2 hardware.  Right now struggling to figure out how to make a pull request.  Reading the instructions doesn't help.

No need to keep this issue open.

dazz100, 2025-02-20:
My version of the firmware/hardware does not allow for detection of 1 LED or 2 LED hardware by the software.
At present, the source code includes a compiler directive.  Selecting the wrong directive is a non-fatal error.  

So an improvement would be to allow the installed firmware to be configured in NVR to run 1 LED or 2 LED software.

xyphro, 2025-04-06:
Thanks Dazz, I'll merge it in!