xyphro/UsbGpib #68: unpack_from requires a buffer

paulwunner, 2024-11-09:
I can't read out anything from the usbgpib adapter. When using the script from #55 without any arguments I get back an error which ends with:

`  File "/home/*user*/anaconda3/lib/python3.12/site-packages/usbtmc/usbtmc.py", line 462, in unpack_bulk_in_header
    msgid, btag, btaginverse = struct.unpack_from('BBBx', data)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: unpack_from requires a buffer of at least 4 bytes for unpacking 4 bytes at offset 0 (actual buffer size is 0)
`
I can however communicate with the GPIB-Device (a Keithley 195A Multimeter) itself without this error.

I am trying this on Ubuntu24

Greetings

jimthouston, 2024-11-11:
It is taking me some time to remember what I was doing in April.  I have not played with usbgpib since then.

I just installed a Ubuntu 24.04.1 LTS and I got the usbgpib_conf.py script to work.  I ran into different errors.
Here are the install steps I used:

```
sudo apt install python3-pip
sudo apt-get install python3-full
python3 -m venv venv
source venv/bin/activate
pip3 install python-usbtmc
pip3 install PyUSB

Add /etc/udev/rules.d/9-usbtmc.rules:
SUBSYSTEM=="usb", ATTR{idVendor}=="03eb", ATTR{idProduct}=="2065", MODE="0660", GROUP="plugdev"
KERNEL=="usbtmc*", MODE="0660", GROUP="plugdev"
```
Running usbgpib_conf.py gave me the correct information and then crashed in libusb_open().  I fixed this by calling inst.close() before exiting the script.

Paul if you can detail how you installed I would try to reproduce and debug the problem you are seeing.

I also tried usbgpib_gui.py and have a new problem which I'm still chasing.  It looks like something has the nicegui event handling.

Jim

paulwunner, 2024-11-11:
Thanks for the answer Jim, I can't remember exactly how I installed python.
However I am in a virtual environment, just like you, and have python-usbtmc and PyUSB installed. I even tried it in a fresh venv with just those 2 packages.
I also added the rules.
Unlike you, I still get the libusb_open() crash the first time I run the script, even though I added the inst.close() into the script.

Most of the time it crashes on the inst.ask() before even printing the !autoid?. However sometimes it prints the !autoid? and even prints the measurement result of my DMM. When this happens it crashes in the inst.pulse() function. So I can read something! Just not the right thing.

Here is the whole error message that pops up most of the time:

```
python3 usbgpib_conf.py
Usage: usbgpib_conf command
Traceback (most recent call last):
  File "/home/*user*/anaconda3/usbgpib_conf.py", line 10, in <module>
    print(cmd, inst.ask(cmd))
               ^^^^^^^^^^^^^
  File "/home/*user*/anaconda3/lib/python3.12/site-packages/usbtmc/usbtmc.py", line 630, in ask
    return self.read(num, encoding)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/*user*/anaconda3/lib/python3.12/site-packages/usbtmc/usbtmc.py", line 613, in read
    return self.read_raw(num).decode(encoding).rstrip('\r\n')
           ^^^^^^^^^^^^^^^^^^
  File "/home/*user*/anaconda3/lib/python3.12/site-packages/usbtmc/usbtmc.py", line 542, in read_raw
    msgid, btag, btaginverse, transfer_size, transfer_attributes, data = self.unpack_dev_dep_resp_header(resp) 
                                                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/*user*/anaconda3/lib/python3.12/site-packages/usbtmc/usbtmc.py", line 466, in unpack_dev_dep_resp_header
    msgid, btag, btaginverse = self.unpack_bulk_in_header(data)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/*user*/anaconda3/lib/python3.12/site-packages/usbtmc/usbtmc.py", line 462, in unpack_bulk_in_header
    msgid, btag, btaginverse = struct.unpack_from('BBBx', data)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: unpack_from requires a buffer of at least 4 bytes for unpacking 4 bytes at offset 0 (actual buffer size is 0)
libusb: debug [libusb_open] open 1.14
Speicherzugriffsfehler (Speicherabzug geschrieben)
```

I'm not really sure how to further describe my situation.

jimthouston, 2024-11-11:
Hi Paul,

Here is the combination which is working for me:
Ubuntu 24.04.1 LTS
python 3.12.3
pyusb 1.2.1
usbtmc 0.8
usbgpib firmware V1.6

I see that there is a V1.7 FW which I have not tried.

Jim

jimthouston, 2024-11-12:
I installed the V1.7 FW and I'm still unable to reproduce the problem.
All I have been testing is running the usbgpib_conf.py.

Paul are you seeing this problem first time every time or might it require accessing the attached gpib device?

paulwunner, 2024-11-12:
Hey Jim,
I guess I  have the V1.7 FW as since my programming no change happened on Github.
The problem happens every time. I'm entirely not able to read out anything from the adapter.

jimthouston, 2024-11-12:
Hi Paul,

I see from the errors that you are using Anaconda as your python package manager.  I have not used it but it looks like it has magic to export an environment to allow reproducing problems. 

If you would do this I will try to reproduce the problem.   I think activating your environment and running this command will create the needed file:
```
conda env export > environment.yml
```

I have memories of github refusing to let me attach a file.  If it doesn't like the .yml file try compressing it with gzip.

Also am I correct in assuming that you are running Ubuntu 24.04.1 LTS on an X86_64 PC?

Jim

paulwunner, 2024-11-20:
Hey Jim, sorry for the delay, I'm somewhat busy.
Here is my environment that you asked for.
[environment.zip](https://github.com/user-attachments/files/17835343/environment.zip)

Yes I am running Ubuntu 24.04.1 LTS on a X86_64 machine.

jimthouston, 2024-11-20:
Hi Paul,

I tried your Anaconda environment.yml and everything is still working correctly here.  I still plan to double check some revisions to see if conda has done what I expect.

I'm trying to imagine anything that we might be doing differently.  I only have one UsbGpib and I suspect that there might be confusion if there was more than one connected. 

Do you have any other software running which might be using the UsbGpib.

You might be able to figure out what is going on by using Wireshark to capture a USB trace.

jimthouston, 2024-11-21:
Hi Paul,

An obvious difference might be the connected GPIB device.  Could you try a different GPIB device?  What make/model is the DMM you are using?

There could be a race between the inst.pulse command and the inst.ask() and something happening on the GPIB bus.  I'm wonder if your DMM is setup to periodically send the value?   

I have been playing with the usbgpib_gui.py script.  I was trying to read back the configuration option I had changed and getting USB EPIPE errors.  I found that the error went away when I changed an inst.write() to an instr.ask().  I had been using the write when I was sending a command where I didn't expect a reply.  When I used the inst,ask() the response was the voltage value from my HP3478A meter.

xyphro, 2025-02-11:
I know I'll make some people upset, but I'd not neccesarily recommend to use usbtmc.py. Having a rough look at it, it does e.g. not seem to have consistent error handling implemented, especially not for the ask function.

jimthouston, 2025-02-12:
Hi Kai,

I don't have much experience with proprietary Visa back ends.  I tried the Agilent libraries when I was having problems with Linux Gpib on an Agilent 82357B.  I was trying to read the calibration data from an HP 3478A meter and had a hard  failure with the open source Linux driver.  When I tried to reproduce the problem with the Agilent Visa libraries I found that it always sent a GPIB sub-address which wasn't supported by the HP 3478A meter.  I reported this bug to Agilent and it was immediately closed because it was obsolete hardware.  I worked with Dave Penkler to fix the Linux 82357 driver.

I was pleased to find the python-usbtmc because it had the pulse() function needed to configure the UsbGpib. It also has the advantage of being portable.  I thought about adding support for control_in to the pyvisa-py.  I went with the python-usbtmc because what I wanted was the quick hack to support the UsbGpib on Linux.

Going back to Paul's issue.  It is still a mystery why the script always works for me and fails first time every time for
Paul.  I'm convinced that difference installing with pip vs. anaconda was not the problem.  I suspect that there was some other process reading the data from the UsbGpib.  My guess is that he has a VISA back end installed that has a daemon probing for new hardware.   I remember the Agilent Libraries did this.

When I get some time, I could hack the scripts I wrote to check if pyvisa is present and has support for control_in() and use it or fallback to using python-usbtmc.  I'm burnt out on proprietary software so I would be looking for someone else to test this.

Jim

xyphro, 2025-02-12:
Do you have means to capture USB traffic? E.g. wireshark or a hw method?

With a trace file and the script which shows what it was intended to do such things can be traced down very well I think.

xyphro, 2025-02-14:
Will this stuff run on a raspberry? Can try to reproduce and capture myself. I just need to run the script from #55, right?

jimthouston, 2025-02-14:
Hi Kai,

The scripts should work fine on Raspberry Pi.  I may give it a try.  I'm sure it will work.

The larger problem is that we have not heard from Paul (the original poster) since November.  I think he got scared off when I suggested tracing the USB bus.  I could help have with Wireshark.

I have NOT been able to reproduce the problem.   So why?  Here are my theories:

1. Software configuration.  I have duplicated the same Ubuntu 24.04 and the same python install, but it could be other software which is interfering.  We don't know if Paul had a proprietary VISA back end installed.
2. Is something happening on the GPIB bus?  Paul mentions a Keithley 195A Multimeter.  I doubt this is the problem because Paul said that the UsbGpib worked to talk to the multimeter.
3. Something wrong with the UsbGpib.   Paul says that he "guesses that he has the 1.7 firmware".  Could he be running the earlier firmware that doesn't support the pulse command mechanism to access the parameters?  Paul, if you see this you could read the firmware back and check the version.
4. I updated the scripts to support multiple UsbGpib devices since this issue.  Maybe Paul has more than one.  I suspect that the old script would just pick the first device.  Maybe it picked a device with old firmware.
5. I wonder how the script behaves if the UsbGpib is in bootloader mode?

I hope Paul will answer now that you are paying attention to this problem.

Jim

xyphro, 2025-02-14:
Thanks for the excellent summary.

Either way I can try if I can reproduce the exact same problem, maybe under enforcement of fw changes (e.g. pulse delay time increase).

I'll keep the ticket open.

jimthouston, 2025-02-15:
> I know I'll make some people upset, but I'd not neccesarily recommend to use usbtmc.py. Having a rough look at it, it does e.g. not seem to have consistent error handling implemented, especially not for the ask function. Why I am saying this in a bit provocative way: In the beginning when making this adapter I debugged for many hours an issue which turned out to be originated in a pyvisa-py implementation, which is also using a "native" user space implementation of Visa. It turned out to be a visa library implementation issue.

Hi Kai,

Python tends to handle errors using exceptions.  It is a cleaner solution than having to check for errors and report them back through all the layers of function calls.  I was in "quick hack" mode when I wrote my scripts so they default to generating stack trace backs.  The usbgpib_conf.py could handle errors better with this change:
```
-usbgpib_conf()     
+try:
+    usbgpib_conf()     
+except Exception as e:
+    print(e)
```
This would give an error like this for a device which is powered off:
```
python3 x.py -s GPIB_22_34231323231351E01261
UsbGpib config for GPIB_22_34231323231351E01261
Device not found [init]
```

The python-usbtmc error handling is probably fine.

A user space implementation raises concerns if multiple processes try to access the same device.   
A kernel driver close() function will always be called when a process exits.   A user space implementation may not get the same chance to clean up when a process exits.   I found that explicit calls to close the instrument made things work better.

Jim

jimthouston, 2025-02-15:
Going back to the original problem

I tried to check out some of the ideas from my earlier post.  Nothing I tried gives the empty buffer which Paul reported.

I experimented with earlier versions of the firmware that didn't support the pulse() based commands to access the parameters.  I got two different failures.  Since the UsbGpib didn't recognize the command it was sent to the connected device.   An HP3478A is happy to respond to an invalid command with the current voltage reading so I got that as the parameter values.  An HP6642A doesn't respond and I got timeout exceptions.

I also tried putting the UsbGpib in boot loader mode.  This gave the "device not found" exception.

I wondered it the problem might be related to the speed of the host relative to the UsbGpib.  I tried sleeping for a few milliseconds between the inst.pulse() and the inst.query().  This just worked.  It read the parameters correctly.

Jim

xyphro, 2025-02-15:
Jim, do you know what timeouts are used for the control transfer in usbtmc.py?

I think you had earlier the suspicion that the pulse indicator delays cause some of the trouble.

Sorry, could not yet spend time on it - busy supporting some guys on my hp3457a oled display project, but it converges...

jimthouston, 2025-02-15:
Hi Kai,

Knowing about the busy wait in the pulse() command was part of the reason I tried adding a sleep between the pulse() and query().  I tried sleeping for a few values starting with 5 milliseconds and trying values up to 2 seconds.  It all just worked.

From a quick look the usbtmc.py uses a timeout of 5 seconds.  It sets the timeout to 5.0 and multiplies this by 1000 to convert to ms when calling ctrl_transfer() and the reads and writes on the bulk end points.

When working on usbgpib_gui.py I got messages in the logs if I scanned for devices at the same time the pulse command was busy waiting.  It was trying to read the LangID value and failing.  Other than the log messages things seemed to work.

Jim

paulwunner, 2025-04-04:
Okay I do as follows:
Connect usbgpib to my pc
Connect usbgpib to HP4195A -> usbgpib blinks slowly
Power on HP4195A -> usbgpib blinks quick and stays on
Run usbgpib_conf.py -> instant response: UsbGpib config for GPIB_26_3423130383135151A121
After a short delay I get the same error
`struct.error: unpack_from requires a buffer of at least 4 bytes for unpacking 4 bytes at offset 0 (actual buffer size is 0)
libusb: debug [libusb_open] open 1.8`
The error traceback leaves the script via the inst.ask(cmd) in display_conf()

If I run it again, I get the same instant response but the error:
`struct.error: unpack_from requires a buffer of at least 4 bytes for unpacking 4 bytes at offset 0 (actual buffer size is 0)`

Paul

xyphro, 2025-04-04:
Thanls Paul, this is extremly useful - will analyze!

xyphro, 2025-04-05:
Hi Paul, 
looking right now at the .zip file. It contains the pcap trace:
Here what I see:
#47: Get Capabilities request => Correctly answered in #48
#49: INDICATOR_PULSE request => Correctly answered in #50
#51: "!autoid?" gets sent as GPIB command
#53: a GPIB read transfer gets executed
**#55: The device replies with a ZERO length package**

So #55 is incorrect - a USBTMC header is not returned by the device. The behavior I see here is exactly the one I would expect from earlier firmware versions. Those had namely no ability to read back the current setting and had another command format for making settings. It is also incompatible with the GUIs which were developed for later firmware versions.

Unfortunately early versions had btw. also no firmware version readout option. The Idea to integrate this only came later (A bad idea to skip that in the beginning).

Can you update to the latest version here?
https://github.com/xyphro/UsbGpib/blob/master/SW/binaries/TestAndMeasurement.bin
    => Click the Download RAW button on the right side
[image]

Best regards,

Kai

xyphro, 2025-04-05:
I hope that's it. I should add a note that the GUI(s) require to have newer FW versions.

jimthouston, 2025-04-05:
There is a chance to know the firmware Paul is running by reading the image
iback from the UsbGpib using the boot loader mode before copying in the new
firmware.

Jim

paulwunner, 2025-04-06:
Okay now I feel kind of ignorant not having tried this before, but HURRAY. With the firmware update it works.
For the curious, here is the stuff that was on the device before the update.

[firmwareOldfromDevice060425.zip](https://github.com/user-attachments/files/19620342/firmwareOldfromDevice060425.zip)

Thanks for the amazing help and again sorry for the suuper late responses.

Kind regards, Paul

closed 2025-04-06 by xyphro