xyphro/UsbGpib #70: Python script for config commands

aleruggeri87, 2024-12-01:
Hi,

I would like to add some custom configuration parameters.
Is it possible to share the python script that generates the cmd_parser constant in miniparser.c?

Thanks,
Alessandro.

dazz100, 2024-12-03:
I am also seeking the python script.  I have just figured out how to change the firmware version number stored on in nvr, but I need to be able to read the values to test my change.

dazz100, 2024-12-09:
I found the py script here:
[https://github.com/xyphro/UsbGpib/discussions/55](url)
I have not yet tried it.

aleruggeri87, 2024-12-10:
Hi,
thanks for pointing out the script.
Actually, I was interested to the script that generates the constant for the command parser on the MCU, not the one that sends commands from the PC side. In the file I mentioned earlier, it’s stated that the parsing constant is generated using a Python script.
My goal is to add new commands, not just modify the replies of the existing ones.

Cheers,
Alessandro.

dazz100, 2024-12-11:
Hi
I am interested in what you plan to do.   
Adding or modifying constants seems to be easy enough. Doing something with those constants could be difficult.

Please keep me updated either here or at my forked repository.

xyphro, 2025-01-11:
[ParserGenerator.zip](https://github.com/user-attachments/files/18387623/ParserGenerator.zip)

Shared herewith without any warantee - use a own risk :-)

closed 2025-01-11 by xyphro