xyphro/UsbGpib #71: REV 2 housing captive nuts

JasonMorris1, 2024-12-07:
I'm finding the captive nut slot for the m3 nuts to be a bit large.  The nut is very loose and starts to spin If I tighten them to much. Could you please reduce the size slightly? Thanks.

xyphro, 2025-01-11:
Hi Jason,

such things are very dependent on the printer used. Changing it for everybody in the right way will be difficult. Also other paramters like the cavity for the PCB can be dependent on the actual printer. The Nut used is a standard DIN934 M3 nut.

With housings printed with JlcPcb, on Bambu P1S, Creality K1, Creality Ender 5 it fits quite well.

I understand your difficulty, but suggest you change the nut cavity yourself to match your actual nut size.

Best regards,

Kai

closed 2025-01-11 by xyphro

xyphro, 2025-01-11:
Note that I will put the Fusion 360 file in a few minutes under housing/REV2/XyphroUsbGpibRev2_housing_part2.f3d

This will enable you and others to do edits yourself.