xyphro/UsbGpib #77: Pyvisa script hangs after 45-60 minutes when talking to two UsbGpib adapters

ned-pcs, 2025-02-05:
I have to do overnight logging of experimental values from two older bench multimeters. I have two UsbGpib adapters using what I believe is the latest firmware. I will check shortly.

The problem is that the scripts will run properly for 45-55 minutes or so and then hang in a way that I haven't been able to catch programmatically.

My Python script is using PyVisa and I have no other back-end drivers installed on my Mac OS 15.2 (Mac Mini M1) system.

The meters are an HP 34401A (in native mode) and a Fluke 8442A.

The script gets readings from both meters every 10 seconds. When it hangs I can't interrupt the program using Ctrl-C but instead have to suspend it and kill it.

I am using `asyncio.timeout()` around the `query()` calls to catch timeouts while waiting to read from the adapters, but these aren't catching the problem:

```
async def read_ac_current():
    async with asyncio.timeout(QUERY_TIMEOUT):
        return float(hp.query('READ?'))

async def read_batt_current():
    async with asyncio.timeout(QUERY_TIMEOUT):
        return float(fluke.query('?'))
```

I will check the FW revision and update the firmware on the adapters shortly.

Beyond that, what can I do to diagnose/fix this issue? I could test under Linux using a Raspberry Pi 5 if that'd help.

ned-pcs, 2025-02-05:
This may be due to the system sleeping. I just turned off system sleep when the display goes to sleep; will see if that helps.

xyphro, 2025-02-11:
Hi!

Something like this immediately triggers my attention. I tested a lot also standby/sleep survival. A hand is absolutely unacceptable and has to be ressolved if it is caused by the adapter.

In case you have a script to try, OS information and which Visa you use please share with me if you can. You can also mail to xyphro at gmail dot com.

Any updates on your investigation are very welcome.

Best regards,

Kai

xyphro, 2025-02-11:
Are you using real pyvisa or pyvisa-py?

pyvisa-py has some issues, like incomplete USBTMC implementation or not fully correct USBTMC error handling.

(I hope nobody complains now, but I had bad experience with it, also outside of the USB-GPIB context)

xyphro, 2025-02-11:
> Beyond that, what can I do to diagnose/fix this issue? I could test under Linux using a Raspberry Pi 5 if that'd help.

The issue is that I cannot reproduce anything under Mac OSX, as my only Mac machine is that old, that I had to install linux on it to make still use of it.

If we can make this reproduceable under any different OS, it would help. E.g. Intel+Windows, Intel+Linux, Raspberry Pi 3..4..5 + linux.

Can't remember, but does Mac OSX have dmesg? See anything related to USB there?
Do you need to unplug the usbgpib to make it "talk" again or is killing the python instance and starting it again all you need?

If you use pyvisa-py, can you try a proper visa [1]+ normal pyvisa?

[1] Any of your prefrence, e.g. NI, R&S. Better go with R&S Visa as that one does not need 1 GByte of harddisc space.

ned-pcs, 2025-02-13:
Sorry I didn't get back here. The problem was indeed with system sleeping. Turning off system sleep got rid of the problem.

closed 2025-02-13 by ned-pcs

xyphro, 2025-02-13:
> Sorry I didn't get back here. The problem was indeed with system sleeping. Turning off system sleep got rid of the problem.

Thanks for update. For self-learning: did you use pyvisa + external visa or pyvisa-py?