xyphro/UsbGpib #78: A collection of small observations and issues

bryanparadis, 2025-02-06:
1. Couldn't really find a good source the USB-C port for Rev 2 but I think DX07S016JA3R1500 looks like it's a match foot print and layout wise.
2. [C7 in V2 BoM ](https://github.com/xyphro/UsbGpib/blob/master/HW/REV2/BOM-xyp-gpib.csv#L5) should be 1uF not 1u
3. C7 in V2 HTML BoM same thing
4. [Component sourcing section](https://github.com/xyphro/UsbGpib?tab=readme-ov-file#component-sourcing) does not match V2 and was a bit confusing to read at first
5. GPIB connector isn't in V2 BoM at all

I created a list on Digikey https://www.digikey.ca/en/mylists/list/QU9XX4NWSD but have not yet built a board so ymmv

bryanparadis, 2025-02-06:
Also wondering if there is information about PCB options and thickness? Going to take a look at the model to see how much room there is. Have never ordered a PCB before so it could get interesting (:

bryanparadis, 2025-02-06:
DIN934 M3 captive nuts and screws could be added to the bom possibly?

bryanparadis, 2025-02-06:
https://www.digikey.ca/en/products/detail/norcomp-inc/111-024-113L001/955147

Will this one work for REV 2?

xyphro, 2025-02-11:
> Couldn't really find a good source the USB-C port for Rev 2 but I think DX07S016JA3R1500 looks like it's a match foot print and layout wise.

A partnumber is included in the BOM csv file. C2936185 from LCSC/JLCPCB.
This is a very common connector, you'll find plenty of the same ones.

> C7 in V2 BoM should be 1uF not 1u
> C7 in V2 HTML BoM same thing

Will delete that HTML BOM soon.

> GPIB connector isn't in V2 BoM at all

For reasons: It can only be mounted by hand. e.g. selective wave soldering is not possible as the SMD components are too close. 
To avoid any issues, I excluded the connector.

> Also wondering if there is information about PCB options and thickness? Going to take a look at the model to see how much room there is. Have never ordered a PCB before so it could get interesting (:

Standard 1.6mm.

Keep your life easy, order at JlcPCB. BOM-xyp-gpib.csv, CPL-xyp-gpib.csv and the Gerber files are all in the REPO. Zip the gerbers, upload it, select PCBA, upload the BOM and CPL and that's all you need to get fully SMD mounted PCB(s) back.
No need to do classical sourcing with Farnell, Digikey, Mouser. If I remember it will cost you ~15 USD for small volumes per piece. Just order 2 screws, 2 nuts and the GPIB connector seperately, print the STL files and you are ready.

> DIN934 M3 captive nuts and M3x20 screws could be added to the bom possibly? I see they are on the Rev 2 readme page.

I'd like to keep the BOM for the PCB only.

> https://www.digikey.ca/en/products/detail/norcomp-inc/111-024-113L001/955147
> Will this one work for REV 2?

Yes sir! This partnumber is written in the schematic.
Most people order that one from AliExpress "get a USB connector "24P Male solder type" from e.g. Aliexpress"

closed 2025-02-11 by xyphro

bryanparadis, 2025-02-12:
> A partnumber is included in the BOM csv file. C2936185 from LCSC/JLCPCB. This is a very common connector, you'll find plenty of the same ones.

Couldn't source or find a substitute by using either number or manufacturer part number. I had to manually search for a part on Digikey that fit the same footprint and pinout. Is LCSC common? I see the JLCPCB has it listed.

> For reasons: It can only be mounted by hand. e.g. selective wave soldering is not possible as the SMD components are too close. To avoid any issues, I excluded the connector.

Yeah process wise this makes sense.

> Standard 1.6mm.

I figured it must be standard after doing some measurements in the 3d model. Luckily I ordered correctly. Thanks for the confirmation.

> Keep your life easy, order at JlcPCB. BOM-xyp-gpib.csv, CPL-xyp-gpib.csv and the Gerber files are all in the REPO. Zip the gerbers, upload it, select PCBA, upload the BOM and CPL and that's all you [...]

That is a very succinct set of instructions. I would do just that but I've got bare PCB in hand now and all the parts I need coming from Digikey.

> I'd like to keep the BOM for the PCB only.

I mean there is a BOM for the PCB and there is a BOM for entirety of the product. Currently the information is split up. My feedback really is it was hard to find everything that I need to put together one of these.

> Yes sir! This partnumber is written in the schematic. Most people order that one from AliExpress "get a USB connector "24P Male solder type" from e.g. Aliexpress"

Aliexpress just takes to long to arrive. Sometimes upwards of 3-4 months if it goes through the UAE on it's way to Canada. I order from there if it's worth the wait only/can't get it from newark/digikey/mouser.

bryanparadis, 2025-02-17:
For anyone that reads this:

The pinout in LCSC schematic is pretty much unreadable. 
[image]

[image]

Also DX07S016JA3R1500 is not a match pinout wise. Trying USB4105-GF-A which does seem to match properly and share the same footprint. I updated my Digikey bom.

xyphro, 2025-02-17:
Hi Bryan, 

this project is made fully for free and everybody is able to make a pull request to improve and I value those a lot. 95% want jlcpcb one stop shop which is provided, I prefer farnell, others local vendors like Reichelt and others again Mouser/Digikey. Maintaining all this information continously to be consistent is difficult for just a hobby project which makes 0 income.

After 6+ years there were only a hand full of pull requests while thousands have built this circuit already.

If you want production quality BOM for your vendor I am happy to add it once I see a pull request.

Best regards,

Kai

bryanparadis, 2025-02-18:
> Hi Bryan,
>
> this project is made fully for free and everybody is able to make a pull request to improve and I value those a lot. 95% want jlcpcb one stop shop which is provided, I prefer farnell, [...]

I like the project but it was hard to figure out how to build to be honest. I may open a PR with improvements after I get a working version on hand. Though the last PR was merged in 2022 and there are 6 open ones.