xyphro/UsbGpib #80: Cannot set autoid slow

rapgenic, 2025-03-19:
Hi, I have built a few of your adapters, they are really great!

I am using one of them with the HP53132A Universal Counter, which seems to open the GPIB interface before being actually able to answer to the `*IDN?` query.

For this reason I wanted to set the `autoid slow` option, however the adapter (two of them actually) completely ignores my query. 

I am able to set the autoid option to either `on` or `off`:

```bash
(venv) $ python usbgpib_conf.py -s HEWLETT-PACKARD_53132A_0_3944 autoid off
!autoid off
(venv) $ python usbgpib_conf.py -s HEWLETT-PACKARD_53132A_0_3944 autoid?
!autoid?
off
```

Then

```bash
(venv) $ python usbgpib_conf.py -s HEWLETT-PACKARD_53132A_0_3944 autoid on
!autoid on
(venv) $ python usbgpib_conf.py -s HEWLETT-PACKARD_53132A_0_3944 autoid?
!autoid?
on
```

But when setting it to slow it still answers `on`:

```bash
(venv) $ python usbgpib_conf.py -s HEWLETT-PACKARD_53132A_0_3944 autoid slow
!autoid slow
(venv) $ python usbgpib_conf.py -s HEWLETT-PACKARD_53132A_0_3944 autoid?
!autoid?
on
```

I have also tried your "official" gui in Windows, with the same result. Note that this happens on two of the adapters that I have built, I can try on more if needed.

This is the full output of the `usbgpib_conf.py` script, for completeness:

```bash
(venv) $ python usbgpib_conf.py -s HEWLETT-PACKARD_53132A_0_3944 
UsbGpib config for HEWLETT-PACKARD_53132A_0_3944
!autoid? on
!term? eoi
!string? normal
!ver? V1.8
```

I'd be really grateful if you could help me understand what I'm doing wrong...

xyphro, 2025-04-06:
Hi!

That's annoying indeed - will check immediately!

Best regards,

Kai

xyphro, 2025-04-06:
Done: I confirm your finding. How can I have missed that one?!

xyphro, 2025-04-06:
It is fixed in attached version. Thanks a lot for reporting it!

[TestAndMeasurement_1.9.zip](https://github.com/user-attachments/files/19620662/TestAndMeasurement_1.9.zip)

I will release it later - there is one other thing outstanding + README.md update.

closed 2025-04-06 by xyphro

X-Microsystems, 2025-04-20:
Although the `slow` setting is now selectable in V1.9, `slower` and `slowest` both set the value to `slow` as well.

```
!ver? V1.9
!autoid? slow

!autoid off
!autoid? off

!autoid on
!autoid? on

!autoid slow
!autoid? slow

!autoid slower
!autoid? slow

!autoid slowest
!autoid? slow

!autoid off
!autoid? off

!autoid slowest
!autoid? slow

!autoid off
!autoid? off

!autoid slower
!autoid? slow

!autoid offtest
!autoid? off

!autoid test
!autoid? off
```

xyphro, 2025-04-20:
Hi!

You are right, I have to modify the parser to allow keywords that start with the same letters. It actually never worked.

Having that said: I got hinted by the usbtmc kernel mode maintainer that usbtmc has a feature to allow setting read ternination directly with VISA or usbtmc via IOCTL.

So working on that and will turn as default for new hw builds the autoid off while not affecting old ones - they will behave as it and the settings will stay.

Saying this just to say: working on it anyway and will share soon an update.

Best regards,

Kai

xyphro, 2025-04-20:
I have fixed this already, however don't want to release it yet as I am busy improving the read termination as mentioned in earlier post. If you have an urgent need let me know and I'll drop here a FW.

xyphro, 2025-04-21:
Check again with the just released version V2.0. I did a deep check of all the options.

rapgenic, 2025-05-06:
Hi Kai!

Thank you for your fix, I could test it only today.

I am afraid it seems I cannot get it working, still :( 

I have updated the version 2.0 from the `binaries` folder, and tried to configure it with the new binary of your GUI. When I set the speed and then click on Restart Adapter, it still shows either ON or OFF.

I am sure that the FW version is correct as the GUI reports V2.0... I'll keep testing to see if I'm doing something wrong...