xyphro/UsbGpib #81: read returns only chuncks

VK2BEA, 2025-04-15:
I'm adapting come C / GTK+ code (to control the HP8753 network analyzer) to use this USBTMC device.
It currently uses the traditional linux gpib driver.

I can do short reads and writes without problem (like "IDN?;"); however if I try to read a large amount of data, I get only small chunks for each read call.
For example, "OUTPLEAS?;" requests the learning string. It is 2630 bytes in length. If I execute a read (after sending "OUTPLEAS") I get back 1012 bytes. I have to do another two reads (returning 1012 and 606 bytes respectively) to get all the data.
Is this expected behaviour or should I be able to get all the data with a single read (as I would expect)?
Since this is binary data, on the linux GPIB driver I must read the first few bytes to determine the number of binary bytes to request on the second call. Here, I'm using the "read(2) with a sufficiently large buffer (10K)"

Michael

xyphro, 2025-04-18:
Hi Michael,

I want to quickly summarize our email discussions here to log it.

Usbgpib uses for read transfer of data bigger than 1012 bytes multiple usbtmc transfers internally. This is described in the [USBTMC specification](https://www.usb.org/document-library/test-measurement-class-specification) file USBTMC_1_00.pdf Figure 4 and in this note #13: "The device may send a Bulk-IN message using multiple transfers, as the data becomes available. This
is illustrated below in Figure 4. This ability is needed because some devices may not have enough
memory to buffer a complete USBTMC message. Another benefit of this ability is that some Hosts
may make use of USBTMC message content as it is delivered."

When using a VISA interface, the visa layer does automatically concatenate those multiple usbtmc transfers together into one returned set of data. VISA does this internally based on the EOM (end of message) bit from the usbtmc header in the usb frames - the last transfered set of data will have EOM set to 1, the other ones will have it set to 0.

The linux usbtmc kernel mode driver does not do this concatenation internally itself, but exposes the EOM bit information belonging to the latest read seperately with an IOCTL call to read it out.

So ideally an implementation directly interfacing to the usbtmc layer without VISA in between does execute reads like this (clumsy pseudocode, excluding accumulation of actual read data and error handling):

```
count_of_databytes_to_read = 1000000 // just an example...
count_of_databytes_read  = 0;
finished = false;
while ( (count_of_databytes_read < count_of_databytes_to_read) && (!finished) )
{
      databytes_read_chunk = start a read of e.g. 4096 bytes
      read bmtransfer field using: ioctl(fd,USBTMC_IOCTL_MSG_IN_ATTR, &res) // res is here a uint8 and returns the EOM bit
      databytes_read += databytes_read_chunk;
      if ( res & 1) // This bitmasking is important, because eom is encoded in bit 0 and bit 1 can be used for other purposes
      {
            finished = true;
      }
}
// databytes_read  contains the actual count of data bytes read
```

For WRITES such a mechanism will not be required.

Let me explain the design choice I made here:
When executing a read over usb, the actual amount of data bytes read from the GPIB device is unknown. However, the USBTMC header encodes the count of read bytes directly in the start of the answer over USB. This means a buffering in memory is required before being able to construct a USB packet and send it back to the host. The MCU has small RAM and 1024 bytes was the choice of memory size here. Independent of the small memory size even e.g. 1 MByte would not be enough because the GPIB device could even return more than 1 MByte. Somewhere a limit has to be drawn. So there is no way to build a bridging device (which has no a priori knowledge of what the device returns) can be done with splitting up to multiple usbtmc transfers.

What would be really nice to have is an usbtmc user mode C module which exposes the usbtmc driver functions in an abstracted way and include functionality like the above. There might be other things which are nice to abstract too,

Best regards,

Kai

xyphro, 2025-04-20:
Hi Michael,

Closing as you have it working. Let me know when further topics pop up. Success in further porting,

Kai

closed 2025-04-20 by xyphro